ihmcrobotics/ihmc-open-robotics-software #129: Valkyrie Leg J1 yaw pivot on 0.9.0

allspawj, 2017-07-14:
| Required Info | |
|---|---|
|Version or commit hash|0.9.0-support|
|Sim or Real Robot|Real Robot|
|Operating System|Ubuntu 14.04|
|JDK|Oracle java version "1.8.0_131"|

#### Steps to reproduce issue
Install 0.9.0-support controller onto the robot. Startup robot, take a few steps

#### Expected behavior
Robot walks reliably

#### Actual behavior
Within a few steps, her leg J1 Yaw will pivot strongly, causing an immediate fall.

#### Contributors with relevant knowledge or expertise (optional)
UOE was having the same problem and I understand @SylvainBertrand was looking into the issue. Was a solution ever found? Do you need more logs?

wxmerkt, 2017-09-11:
Any updates on this @SylvainBertrand, @dljsjr, or @nmertins?

nmertins, 2017-09-12:
I believe this issue has been fixed on the val-develop branch since the 0.9.0-support release. We haven't made a new release though because we don't have hardware to validate it on. Unfortunately we're not sure when that will be either. You can try pulling from val-develop, but you'll also need to switch to the develop branch of ihmc_ros_core because the ROS messages have been updated also.