ihmcrobotics/ihmc-open-robotics-software #130: Missing instructions for Valkyrie Maven setup

tkoolen, 2017-08-21:
https://github.com/ihmcrobotics/ihmc-open-robotics-software#using-ihmc-open-robotics-software-jar-releases-with-mavengradle:

> Instructions for adding the Maven repository and identifying the artifacts can also be found on Bintray for each package.

This is currently not true for Valkyrie, see https://bintray.com/ihmcrobotics/maven-release/Valkyrie#, unless you're referring to the dependency snippet. I was expecting a non-empty readme.

tkoolen, 2017-08-21:
@dljsjr told me that I should be using the source build (currently `0.9-support`) for Valkyrie, instead of Maven. In general, could the instructions for setting up the Valkyrie software be made a bit clearer? Thanks!

dljsjr, 2017-08-23:
We should probably have a standalone set of instructions in the docs for Valkyrie since that's a use-case distinct from general open source usage and definitely distinct from importing our software as a dependency in a Maven build. Thanks for the heads up on this.

tkoolen, 2017-08-23:
Here are my working notes: https://docs.google.com/document/d/1YUi9qJ8nmFPg0sZgWqOUFnBXqz4EHWJPe8aY3wy0CKs/edit?usp=sharing

Note that I haven't gotten it working yet. BTW, just to confirm: I should be using Oracle's JVM on link right, not OpenJDK?

dljsjr, 2017-08-24:
Any version of JDK 8 is suitable as a runtime. But for compiling source code on Ubuntu 14.04 you'll need Oracle JDK 8, because OpenJDK 8 does not have an OpenJFX package on 14.04. So we recommend that OCU computers run Oracle JDK 8.

calvertdw, 2018-10-18:
We have overhauled all of our READMEs and are no longer supporting 0.9, so closing this.

closed 2018-10-18 by calvertdw