ihmcrobotics/ihmc-open-robotics-software #131: RemoteValkyrieVisualizer NullPointerException

tkoolen, 2017-08-28:
| Required Info | |
|---|---|
|Version or commit hash| 0.9-support, `2c47c9a9bd999e7811038e99c3888683f9973a2a` |
|Sim or Real Robot| RemoteValkyrieVisualizer, connecting to Valkyrie B |
|Operating System| Ubuntu 14.04 |
|JDK| java version "1.8.0_144", Java(TM) SE Runtime Environment (build 1.8.0_144-b01), Java HotSpot(TM) 64-Bit Server VM (build 25.144-b01, mixed mode), Oracle|

#### Steps to reproduce issue
1. Run IHMC controller on Valkyrie.
1. Start `RemoteValkyrieVisualizer`
1. Double click on controller when it comes up

#### Expected behavior
SCS window comes up.

#### Actual behavior
Sometimes, SCS window does not come up. Console output:
```
[INFO] (NetworkParameters.java:38): Looking for network parameters in network parameters file at /home/val/.ihmc/IHMCNetworkParameters.ini
[INFO] (NetworkParameters.java:42): Found Network parameters file at /home/val/.ihmc/IHMCNetworkParameters.ini
[INFO] (NetworkParameters.java:69): Looking for network parameters in environment variables
[INFO] (NetworkParameters.java:70): Environment variables will override entries in the network parameters file.
Connecting to host 10.185.0.100
Loading robot model from: 'models/val_description/sdf/valkyrie_sim.sdf'
[INFO] (LogSessionCallback.java:258): Listening on interface name:eth0 (eth0)
[INFO] (LogSessionCallback.java:258): Listening on interface name:lo (lo)
[INFO] (LogSessionDisplay.java:221): Showing
[INFO] (LogSessionDisplay.java:297): ValkyrieRosControlController came online
Requesting handshake
[INFO] (KryoObjectClient.java:130): Success! Connected KryoClient to SCS at ip /10.185.0.20 on port 56322
[INFO] Found graphics adapter: us.ihmc.jMonkeyEngineToolkit.jme.JMEGraphics3DAdapter
Exception in thread "main" java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at us.ihmc.simulationconstructionset.gui.EventDispatchThreadHelper.invokeAndWait(EventDispatchThreadHelper.java:29)
	at us.ihmc.simulationconstructionset.SimulationConstructionSet.<init>(SimulationConstructionSet.java:458)
	at us.ihmc.simulationconstructionset.SimulationConstructionSet.<init>(SimulationConstructionSet.java:372)
	at us.ihmc.simulationconstructionset.SimulationConstructionSet.<init>(SimulationConstructionSet.java:367)
	at us.ihmc.robotDataVisualizer.visualizer.SCSVisualizer.start(SCSVisualizer.java:219)
	at us.ihmc.robotDataLogger.YoVariableClient.start(YoVariableClient.java:229)
	at us.ihmc.robotDataLogger.YoVariableClient.start(YoVariableClient.java:208)
	at us.ihmc.valkyrie.visualizer.RemoteValkyrieVisualizer.<init>(RemoteValkyrieVisualizer.java:48)
	at us.ihmc.valkyrie.visualizer.RemoteValkyrieWalkingVisualizer.<init>(RemoteValkyrieWalkingVisualizer.java:9)
	at us.ihmc.valkyrie.visualizer.RemoteValkyrieVisualizer.main(RemoteValkyrieVisualizer.java:102)
Caused by: java.lang.reflect.InvocationTargetException
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1321)
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1296)
	at javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1348)
	at us.ihmc.simulationconstructionset.gui.EventDispatchThreadHelper.invokeAndWait(EventDispatchThreadHelper.java:25)
	... 9 more
Caused by: java.lang.NullPointerException
	at javax.swing.ImageIcon.<init>(ImageIcon.java:240)
	at us.ihmc.simulationconstructionset.gui.actions.dialogActions.AbstractActionTools.loadImageIconUsingInputStream(AbstractActionTools.java:46)
	at us.ihmc.simulationconstructionset.gui.actions.dialogActions.AbstractActionTools.setupIconButton(AbstractActionTools.java:21)
	at us.ihmc.simulationconstructionset.gui.actions.StepForwardAction.<init>(StepForwardAction.java:27)
	at us.ihmc.simulationconstructionset.gui.StandardGUIActions.createMainWindowActions(StandardGUIActions.java:217)
	at us.ihmc.simulationconstructionset.gui.StandardSimulationGUI.initGUI(StandardSimulationGUI.java:750)
	at us.ihmc.simulationconstructionset.gui.StandardSimulationGUI.access$22(StandardSimulationGUI.java:550)
	at us.ihmc.simulationconstructionset.gui.StandardSimulationGUI$4.run(StandardSimulationGUI.java:505)
	at us.ihmc.simulationconstructionset.gui.EventDispatchThreadHelper.invokeAndWait(EventDispatchThreadHelper.java:19)
	at us.ihmc.simulationconstructionset.gui.StandardSimulationGUI.setup(StandardSimulationGUI.java:500)
	at us.ihmc.simulationconstructionset.SimulationConstructionSet.createGUI(SimulationConstructionSet.java:2084)
	at us.ihmc.simulationconstructionset.SimulationConstructionSet.access$1(SimulationConstructionSet.java:2052)
	at us.ihmc.simulationconstructionset.SimulationConstructionSet$2.run(SimulationConstructionSet.java:463)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

#### Contributors with relevant knowledge or expertise (optional)

dljsjr, 2017-08-31:
We're not sure what causes this but we have this problem ourselves sometimes. It seems specific to Eclipse. The output directory isn't always consistent and so sometimes graphical resources just… aren't there for a few minutes. Re-running a few times or doing a rebuild often solves the problem.

dljsjr, 2018-08-01:
We haven't seen this in a veeeeeeery long time (we're also on newer versions of Eclipse with much better Gradle integration) so I'm going to close this.

closed 2018-08-01 by dljsjr