ihmcrobotics/ihmc-open-robotics-software #133: Compile error due to mismatch of Euclid version

ninja777, 2018-01-26:
| Required Info | |
|---|---|
|Version or commit hash|RC-0.10|
|Sim or Real Robot|N/A|
|Operating System|Ubuntu 14.04|
|JDK|1.8.0_161|

#### Steps to reproduce issue

1. clear gradle cache 
2. checkout RC-0.10 tag 
3. Import project in eclipse or do a gradle refresh if projects are already setup

#### Expected behavior

Code should compile without any error

#### Actual behavior

Several errors are seen due to missing library/classes

#### Analysis
`ihmc-robotics-toolkit` depends on euclid 0.7.5. This is a dependency of several other projects, `ascell` for an example. `ascell` downloads euclid 0.7.9 in it's project dependencies causing the following error `The type us.ihmc.euclid.referenceFrame.FrameTuple3D cannot be resolved. It is indirectly referenced from required .class files` at line 363 of `us.ihmc.steppr.logProcessor.StepprSensorPostProcessor` Several such errors are seen across multiple projects. 

#### Contributors with relevant knowledge or expertise (optional)

dljsjr, 2018-01-26:
@ninja777 I'm not sure who created the RC-0.10 tag. If you need to use build 0.10 I would suggest using the release/0.10 branch.

ninja777, 2018-01-26:
Thanks for the response. I tried with release/0.10.0 and val-develop branch and had similar results. The following changes worked for me, I'm sure it could be easier than what is mentioned below but I am not sure what is the right way.

1. Cloned `ihmc-open-robotics-software` in $HOME
2. Cloned `ihmc-path-planning` and `robot-environment-awareness` inside `$HOME/ihmc-open-robotics-software` (I know this is bad, but I was looking for a quick fix. Any recommendations on how to add these repo so that the argument `use_local_build=true` would work in the last step are appreciated). 
3. Modified gradle.properties in the `$HOME/ihmc-open-robotics-software` directory as follows
```
isProjectGroup = true
hyphenatedName = ihmc-open-robotics-software
pascalCasedName = IHMCOpenRoboticsSoftware
publishMode = SNAPSHOT
groupDependencyVersion = release/0.10.0
depthFromWorkspaceDirectory = 1
includeBuildsFromWorkspace = true
excludeFromCompositeBuild = false
```
4. Modified `ihmc_valkyrie_ros/configurations/api.yaml` from branch release/0.10.0 as follows:
```
!ihmc_ros_java_adapter
programArgs: []
vmConfig:
  heapSize: 4g
  mainMethod: us.ihmc.valkyrie.ValkyrieROSNetworkProcessor
dependencies:
- us.ihmc:valkyrie:0.11.0
```
It is strange but release/0.10.0 doesn't work here.

5. Finally launched the simulator
`roslaunch srcsim unique.launch  grasping_init:=false use_local_build:=true`

dljsjr, 2018-01-26:
Ahh okay, I didn't realize you were using the ROS Java Bridge.

That has not been fully updated to our newest build system. It's on the roadmap for 0.12.0. That is probably what is causing complications (and also what is forcing you to have to clone REA and Path Planning inside of Open Robotics).

ninja777, 2018-01-26:
If I don't clone those repos, gradle downloads their SNAPSHOT versions with euclid 0.7.9 even I specify groupDependencyVersion to be release/0.10.0 and I end up with errors mentioned in the details of this issue.

dljsjr, 2018-01-26:
Yes they need to be cloned (for the 0.10.0 branch anyway) but the ROS Java Adapter hasn't been fully updated to handle that layout.

If you can use the 0.11 release (which is a full release with tag) I would suggest trying that but you will have to manually rebuild `ihmc_msgs`. Open Robotics is self-contained on 0.11 and the ROS Java Bridge should be able to handle that. The ROS libraries have not been maintained since the SRC due to logistical issues but they are back in the release pipeline for 0.12 as I mentioned before.

0.10 release was never finalized as you may see, it had some blocking bugs specific to the Valkyrie hardware platform that we weren't able to fix before having to send the robot back to JSC. There's a good chance that 0.10 gets abandoned because of this, and the next Val supported release becomes 0.12 or 0.13 (depending on timeline.

I suggest using 0.11 in the meantime if you can.

ninja777, 2018-01-26:
Sure, that is good to know. Based on commit messages I assumed 0.11 was for atlas only. I'll move to 0.11 right away, thanks :+1:

dljsjr, 2018-01-26:
Hardware-wise it is only verified on Atlas, all simulations should still work though.

closed 2018-01-26 by ninja777