ihmcrobotics/ihmc-open-robotics-software #134: Valkyrie Deploy / Compilation fails with Footstep Planner (dependency mismatch?)

edbot, 2018-01-31:
| Required Info | |
|---|---|
|Version or commit hash|release/0.10.0|
|Sim or Real Robot|Real Hardware - Console Computer|
|Operating System|Ubuntu 14.04|
|JDK|Oracle, 1.8.0_161|

#### Steps to reproduce issue

```
cd valkyrie
gradle clean deploy
```

#### Expected behavior
Deploy code to robot.

#### Actual behavior
```
> Task :ihmc-footstep-planning:compileJava FAILED
/home/edbot/ihmc-workspace/ihmc-open-robotics-software/ihmc-footstep-planning/src/main/java/us/ihmc/footstepPlanning/graphSearch/footstepSnapping/PlanarRegionBaseOfCliffAvoider.java:102: error: cannot find symbol
         shiftVectorInSoleFrame.sub(highestLineSegmentInSoleFrame.getFirstEndpointCopy());
                                                                 ^
  symbol:   method getFirstEndpointCopy()
  location: variable highestLineSegmentInSoleFrame of type LineSegment2D
/home/edbot/ihmc-workspace/ihmc-open-robotics-software/ihmc-footstep-planning/src/main/java/us/ihmc/footstepPlanning/graphSearch/footstepSnapping/PlanarRegionBaseOfCliffAvoider.java:110: error: cannot access FrameTuple2D
            yoShiftVectorInSoleFrame.set(shiftVectorInSoleFrame.getX(), shiftVectorInSoleFrame.getY(), 0.0);
                                    ^
  class file for us.ihmc.euclid.referenceFrame.FrameTuple2D not found
/home/edbot/ihmc-workspace/ihmc-open-robotics-software/ihmc-footstep-planning/src/main/java/us/ihmc/footstepPlanning/graphSearch/planners/VisibilityGraphWithAStarPlanner.java:198: error: incompatible types: List<Point3DReadOnly> cannot be converted to List<Point3D>
            List<Point3D> path = navigableRegionsManager.calculateBodyPath(startPos, goalPos);
                                                                          ^
/home/edbot/ihmc-workspace/ihmc-open-robotics-software/ihmc-footstep-planning/src/main/java/us/ihmc/footstepPlanning/graphSearch/planners/VisibilityGraphWithAStarPlanner.java:291: error: cannot access FrameTuple3D
            bodyPathPoints.get(i).set(projectedPoint);
                                 ^
  class file for us.ihmc.euclid.referenceFrame.FrameTuple3D not found
/home/edbot/ihmc-workspace/ihmc-open-robotics-software/ihmc-footstep-planning/src/main/java/us/ihmc/footstepPlanning/graphSearch/stepCost/DistanceAndYawBasedCost.java:43: error: cannot find symbol
      return midFootPoint.getPoint();
                         ^
  symbol:   method getPoint()
  location: variable midFootPoint of type FramePoint2D
/home/edbot/ihmc-workspace/ihmc-open-robotics-software/ihmc-footstep-planning/src/main/java/us/ihmc/footstepPlanning/graphSearch/stepCost/BipedalStepAdjustmentCostCalculator.java:157: error: incompatible types: FrameVector3D cannot be converted to FrameTuple3D<?,?>
      frameVectorToPack.setWithoutChecks(frameTuple);
                                         ^
/home/edbot/ihmc-workspace/ihmc-open-robotics-software/ihmc-footstep-planning/src/main/java/us/ihmc/footstepPlanning/graphSearch/stepCost/BipedalStepAdjustmentCostCalculator.java:173: error: incompatible types: FrameVector2D cannot be converted to FrameTuple2D<?,?>
         vectorToPack.setWithoutChecks(frameTuple2d);
                                       ^
Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
7 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':ihmc-footstep-planning:compileJava'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org
```

#### Contributors with relevant knowledge or expertise (optional)
@dljsjr @SylvainBertrand 

cc: @wxmerkt @VladimirIvan

ninja777, 2018-01-31:
This is similar to #133 . It has a little more analysis of whats going wrong. The dependency mismatch here is Euclid version.

edbot, 2018-01-31:
Thank you very much for the response and the pointer @ninja777! Did you manage to resolve it with the manual build?

dljsjr, 2018-01-31:
I did some investigating in to this last night and I believe the problem is a transient dependency, on 0.10 the API-compatible snapshots of Footstep Planning and Robot Environment Awareness have fallen out of scope (I confirmed with the build folks that we recently started to preserve only around ~1mo of snapshots due to a disk issue we are trying to track down), so when 0.10 tries to pulling snapshots of REA and Footstep Planning it gets the latest version. This pulls in a newer version of Euclid, which is why Path Planning (which uses Footstep Planning) gets Euclid-based API errors. It's a transitive issue.

I'm putting together instructions on how to resolve this, I'll update this issue once they're completed.

ninja777, 2018-01-31:
@edbot , yes I did. but it is painful. Unless you are as desperate as I was, I would not recommend doing it and waiting for the fix. It is explained in the comments of [same issue](#133 ). One additional change that I missed out is, modified all build.gradle files that have euclid-core dependency and replaced that euclid-core compileGroup with `compile group: 'us.ihmc', name: 'euclid', version: '0.7.4'`

dljsjr, 2018-01-31:
@ninja777 @edbot attached is an .md outlining a workaround to continue using `release/0.10.0`:

[0_10_fix_instructions.md.txt](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/1683524/0_10_fix_instructions.md.txt)

edbot, 2018-01-31:
Thanks. That worked. Deployed and tested up to receiving ROS communication and running the remote visualiser. We'll test the controller with motor power tomorrow and close the issue if everything runs (smoothly).

wxmerkt, 2018-02-01:
Deployment worked, however, we are seeing oscillatory behaviour when commanding any kind of upper body motion (single arm only or whole body trajectory) - video on Slack.

dljsjr, 2018-02-01:
@wxmerkt thanks for the update, can you open a new issue for the torso oscillations so we can have record of it for when we get a robot back on site? Closing this one.

closed 2018-02-01 by dljsjr