ihmcrobotics/ihmc-open-robotics-software #135: Ros Java bridge fails to load 0.9.0 controllers

ninja777, 2018-01-31:
| Required Info | |
|---|---|
|Version or commit hash|0.9.0 / 0.9-support / 0.9-src-finals|
|Sim or Real Robot|Simulation|
|Operating System|Ubuntu 14.04|
|JDK|Oracle 1.8.0_161|

#### Steps to reproduce issue

1. Checkout any of the 0.9x tag/branch
2. Checkout appropriate versions of ihmc_ros_core(0.9.2), ihmc-ros-control(0.5.0), and ihmc_valkyrie_ros(0.9.0). catkin_make and source devel/setup.bash
3. Run the deployLocal task in Valkyrie project to create ~/valkyrie folder with the controller lib
4. Run srcsim `roslaunch srcsim unique.launch` 

#### Expected behavior

After the gradle dependencies are setup - in subsequent launches, the controllers should be loaded and the robot should be standing in simulation.

#### Actual behavior

Error related to version mismatch of ValkyrieController is seen on the console. 
```
Cannot find class us/ihmc/valkyrieRosControl/ValkyrieRosControlController
[ERROR] [1517424500.274393224, 5.561000000]: us.ihmc.valkyrieRosControl.ValkyrieRosControlController does not extend us.ihmc.rosControl.wholeRobot.IHMCWholeRobotControlJavaBridge
[ERROR] [1517424500.274461696, 5.561000000]: Initializing controller 'ihmc_valkyrie_control_java_bridge' failed
[ INFO] [1517424500.274527117, 5.561000000]: Stopping VM
Exception in thread "main" java.lang.UnsupportedClassVersionError: us/ihmc/valkyrieRosControl/ValkyrieRosControlController : Unsupported major.minor version 52.0
```

When we replace the valkyrie folder with the one generated from 0.8.2, it works fine. (with the older ihmc_msgs)

#### More details
We are in the process of creating public docker images for 0.8.2 and 0.9.0 version of controllers for srcsim so as to facilitate switching between the versions. We do not intend to make any modifications to the ihmc-open-robotics-software:0.9.0. If providing a tarball of working controllers for 0.9.0 is easier than providing a fix/workaround, that would work as well. Thank you.

#### Contributors with relevant knowledge or expertise (optional)

dljsjr, 2018-01-31:
@ninja777 "Unsupported major.minor version 52.0" is a really crappy message that means you're trying to run Java code compiled using a Java 8 JDK in a Java 7 JRE/JVM. Triple check your environment.

ninja777, 2018-01-31:
Thank you. I should have googled the error first.
It was the environment. Though update-alternatives and JAVA_HOME were pointing to JDK8, `/usr/lib/jvm/default-java` was set to openjdk-7.

closed 2018-01-31 by ninja777

dljsjr, 2018-01-31:
Haha no worries, that's one of the least useful error messages that Java emits so I'm familiar with the confusion.