ihmcrobotics/ihmc-open-robotics-software #138: Sending StopAllTrajectories raises the pelvis height

wxmerkt, 2018-07-23:
| Required Info | |
|---|---|
|Version or commit hash|release/0.10.0|
|Sim or Real Robot|Real, Valkyrie Unit D|
|Operating System|Ubuntu 16.04|
|JDK|Oracle JDK|

#### Steps to reproduce issue
1. Sent a ``StopAllTrajectories`` message
#### Expected behavior
1. Robot stops trajectories.

#### Actual behavior
1. Robot stops trajectories **and** raises the pelvis. Sending multiple stop-messages even when no trajectory is set continues to increase the pelvis height.

#### Contributors with relevant knowledge or expertise (optional)
@dljsjr @SylvainBertrand

wxmerkt, 2018-07-24:
No, tested last night and apparently only on robot. We noticed larger than before z-drift (at least perceived, I dont have hard numbers on this) - walking a few steps increased z by 90cm (the issue Henrique showed a video/log off in Slack). We also noticed at the end of every whole-body trajectory, the robot raises the pelvis - as if we have a measured vs planned of last waypoint problem again.

dljsjr, 2018-07-24:
I was able to get someone back in Pensacola to run some tests… it looks like in the newest version of the controller the issue with the pelvis height is gone (because we now control pelvis height slightly differently) but the issue manifests with chest orientation. We'll keep this issue updated as we investigate.

wxmerkt, 2018-07-24:
Thanks, much appreciated.

georgwi, 2018-07-27:
I ran out Valkyrie (B) and could not see the increase in pelvis height. Instead, like Doug said I can see the chest pitching up on every message. This is because on a stop all trajectory message the desired chest orientation is set to the current and there seems to be some tracking error that causes the drift.

Since we can not reproduce the height drift right now and Doug is working on a new release that should have ROS1 support we would like to wait and see if that release fixed the issue for you. Is all right?

We will fix the chest drift that we observed here with the stop all trajectory messages on that release!

wxmerkt, 2018-07-27:
Cool, is the ETA for that as with 0.12 late summer?

georgwi, 2018-07-27:
Not sure @dljsjr sounded like he was getting close?

dljsjr, 2018-07-27:
We don't have a firm date yet, we're trying to sort it out. The next release will be some time before the end of summer though.

closed 2023-12-30 by calvertdw