ihmcrobotics/ihmc-open-robotics-software #139: Hybrid taskspace-jointspace messages are not built correctly

wxmerkt, 2018-07-23:
| Required Info | |
|---|---|
|Version or commit hash|release/0.10.0|

#### Steps to reproduce issue
Have a look at https://github.com/ihmcrobotics/ihmc_ros_core/blob/release/0.10.0/ihmc_msgs/msg/HandHybridJointspaceTaskspaceTrajectoryRosMessage.msg - the message contains neither joint nor task space components. The only hybrid message to do so is ``Chest``.

#### Contributors with relevant knowledge or expertise (optional)
@dljsjr

dljsjr, 2018-07-24:
Hey @wxmerkt!

Thanks for the report. This is fixed in the upcoming 0.12 release, but it probably won't be back ported to 0.10.

dljsjr, 2018-07-24:
This is the release that switches to ROS 2 and leverages the ROS 1 Bridge package for folks that still need to use ROS 1. I'm in Houston this week testing on their robot to make sure that we have the packaging and ansible playbooks sorted out then we'll be doing acceptance tests on our robot in Pensacola when I get back. We don't have a hard ETA on 0.12 but we're aiming to have it out this summer. It's a pretty large release with a lot of changes under the hood so we want to make sure it's well tested.

wxmerkt, 2018-07-24:
Okay, we are happy to switch once it's well tested.

closed 2023-12-30 by calvertdw