ihmcrobotics/ihmc-open-robotics-software #144: REA config changes are merged into `val-develop`

calvertdw, 2018-10-18:
Steven Jens Jorgensen [9:31 AM]
Hey Duncan, I just want to let you know that the latest `val-develop` still has the problem of placing the config file as a `~/.ihmc/Configuration` file instead of being located at `~/.ihmc/Configuration/defaultREAModuleConfiguration.txt` which prevents the REA Module from being launched at networkProcessor startup. The relevant lines are:

https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/develop/valkyrie/build.gradle#L198

https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/develop/valkyrie/build.gradle#L239 

https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/07aea6c5363f5eaa6bc7dfacc1418be0869cdbac/ihmc-avatar-interfaces/src/main/java/us/ihmc/avatar/networkProcessor/DRCNetworkProcessor.java#L172

stevenjj, 2018-10-18:
Also,
https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/val-develop/valkyrie/build.gradle#L239

https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/develop/valkyrie/build.gradle#L239

dljsjr, 2018-10-31:
@stevenjj I just pushed up a possible fix on `val-develop` lemme know if it helps

dljsjr, 2018-11-01:
@stevenjj Could be, let me know

stevenjj, 2018-11-01:
Alright. The path issue seems to be fixed now as we can launch the REA module on Network processor startup by setting this boolean in `ValkyrieNetworkProcessor.java` to true.
```
networkModuleParams.enableRobotEnvironmentAwerenessModule(true);
````

Also, the `LidarBasedREAStandaloneLauncher.java` still works as before. But I don't think it actually loads the config file. I changed the folder name from `~/valkyrie/Configurations` to `'~/valkyrie/stale_Configurations`. I also changed 
````
private static final String MODULE_CONFIGURATION_FILE_NAME = "./Configurations/defaultREAModuleConfiguration.txt";
````
to 
````
   private static final String MODULE_CONFIGURATION_FILE_NAME = "./Configurations/stale_defaultREAModuleConfiguration.txt";
````
and this standalone launcher still worked. 

In any case, the path issue is ok now from deployment standpoint. I don't know what's up with the stand alone launcher. It's working when it shouldn't be.

dljsjr, 2018-11-01:
@stevenjj those paths aren't relative to home directory (the ones for the standalone launcher), they're relative to the working directory. So if you're launching from Eclipse it'll be relative to the `robot-environment-awareness` project.

The property file not existing is only an issue if you attempt to save new properties. If you attempt to load a property and the file doesn't exist it'll use default values.