ihmcrobotics/ihmc-open-robotics-software #146: Footstep planner can step on own feet with Valkyrie

calvertdw, 2018-10-18:


stevenjj, 2018-10-22:
Hey Stephen,
This only happens on the real robot and in gazebo. In SCS, it doesn't show up. Are there parameters we can change to take less conservative steps or increase foot distance with the planner?

stevenjj, 2018-10-22:
Also, the footstep plans themselves look ok. It's during execution that she sometimes steps on her feet. These are particularly evident on consecutive footsteps that have some yaw angle differences

stephenmcc, 2018-10-22:
Gotcha. If the footstep plan looks good then it is a problem with the controller - either the swing foot has bad tracking or the controller is adjusting the footstep too much. Would it be possible to get a log of this happening on the real robot?

stevenjj, 2018-10-22:
That makes sense. 
Alright, sure thing. We're about to get the robot walking again after some CoP offset changes, so we'll send you logs when we can walk.  Thanks for being prompt about this.

closed 2023-12-30 by calvertdw